bambulab/BambuStudio #6987: H2D Dev Mode can't be used with Bambu Studio PR Artifacts

logikal, 2025-05-21:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Built from source by yourself or others

### OS version

macos 15.3.2

### Printer

BambuLab H2D

### How to reproduce

1. Place an H2D in Developer Mode
2. Confirm with the officially released package that you can see the H2D in dev mode. This should work.
3. Close Bambu Studio
4. Download a PR build from a GitHub action in this repo. e.g. https://github.com/bambulab/BambuStudio/actions/runs/15152808133/job/42604329727
5. Unzip the artifact.
6. Browse to $UNZIP_DIR/Users/runner/work/BambuStudio/BambuStudio/install_dir/bin/
7. Open BambuStudio.app
8. Open System Settings -> Privacy & Security
9. Click "Open Anyway" next to "BambuStudio.app was blocked to protect your Mac"
10. Click "Open Anyway" on the popup.
11. Authenticate to allow the app to open.
12. Download the Network Plugin automatically
13. Click OK once the Network Plugin download completes.
14. Browse to Device tab

The H2D in Developer Mode _should_ show up here, but doesn't.

I've also attempted this on a windows 11 machine with the same results. 

Ticket has been opened, #US250506028001. Logs have been uploaded to the ticket.

As near as I can tell, H2D Developer Mode _only_ works with the official signed releases of Bambu Studio, and not GitHub Action built artifacts from PRs on this very repo. 

This severely hampers the ability to contribute to Bambu Studio for H2D printers.

### Actual results

The H2D doesn't appear in the list of devices ever. 
If you are logged in, you can see other devices that are available via the cloud, but not an H2D in Developer Mode.

### Expected results

The Developer Mode H2D should show up in the device list, just like it does in the offcially packaged releases.

### Project file & Debug log uploads

[3DBenchy.3mf.zip](https://github.com/user-attachments/files/20360952/3DBenchy.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

logikal, 2025-05-21:
Attempting to add the H2D via "Bind with Access Code" also fails silently.

logikal, 2025-05-21:
My network "sees" the autodiscovery broadcasts, but it looks like the network plugin or BambuStudio ignores it when using a PR-built artifact.

```
> sudo ngrep -d any -Wbyline port 1990 or port 2021 or port 8883 or port 900 or port 322 or port 6000 or host 192.168.1.40
U 192.168.1.40:2021 -> 255.255.255.255:2021 #1
NOTIFY * HTTP/1.1.
Host: 239.255.255.250:1990.
Server: UPnP/1.0.
Location: 192.168.1.40.
NT: urn:bambulab-com:device:3dprinter:1.
NTS: ssdp:alive.
USN: REDACTED
Cache-Control: max-age=1800.
DevModel.bambu.com: O1D.
DevName.bambu.com: H2D.
DevConnect.bambu.com: lan.
DevBind.bambu.com: free.
Devseclink.bambu.com: secure.
DevInf.bambu.com: wlan0.
DevVersion.bambu.com: 01.01.01.00.
DevCap.bambu.com: 1.
```

RyanHir, 2025-05-21:
This behavior is also seen with a P1S running FW 01.07.00.00 in LAN mode with a Mac. This printer FW version is prior to the addition of Developer mode so this leads me to think this is not specific to printers being in Dev mode, and instead just being a LAN mode issue.

RyanHir, 2025-05-21:
This might be a MacOS specific bug. I just tested the same PR as listed above on Windows and the autodiscovery worked in finding my LAN mode printer. Perhaps there is an entitlement added during the release process for the MacOS builds to enable network discovery?

walterwongbbl, 2025-05-22:
Hi. @logikal @RyanHir 
We are confirming this issue now, thank you very much.

logikal, 2025-05-22:
Thanks, @walterwongbbl ! 

Let me know if there's any other data (logs, videos, screenshots, ngrep/wireshark pcaps, etc) that can help track down and fix the issue!

walterwongbbl, 2025-05-26:
Hi. @logikal 
We really need your log information, as our local team cannot reproduce this issue. 
Please change the log level to trace and reproduce the issue again. Then upload the log file.

[image]

[image]

logikal, 2025-05-27:
Steps taken:

1. Deleted ~/Library/Application Support/BambuStudioInternal
2. Downloaded the zip from this Action: https://github.com/bambulab/BambuStudio/actions/runs/15152808133/job/42604329727
3. Extract the .zip
4. Run the .app from the extracted zip folder in ~/Downloads
4. Attempt to add the H2D via "Bind with Access Code"
5. Enter IP Address and Access Code
6. Enter SN and Model
7. Nothing happens.

I've uploaded [Bambu Studio logs as a zip](https://github.com/user-attachments/files/20447328/H2D.Devmode.on.macos.zip) here and in the ticket, and the output of the macos Console app logs filtered to 'Bambu' to ticket US250506028001.

One thing I noticed is that the Internal Build of BambuStudio never prompts or shows up in the settings for Privacy and Security -> Local Network. Should "BambuStudioInternal" show up here?

[image]

logikal, 2025-05-27:
I've confirmed that's the problem!

MacOS doesn't honor requests/entitlements for access to the Local Network from unsigned code applications.

There's a workaround: Applications run from the terminal are allowed access to the Local Network.

If I run the binary inside the .app manually, access to the network is allowed and I can see my H2D in Dev/LAN mode.

[image]

Bambulab should sign their PR builds to prevent this issue.

lanewei120, 2025-05-27:
@logikal 
thanks for your hard work to identify the root cause

as the signing process needs the Apple developer account of our office
we can only ensure the appimage published by us been signed

currently the image built from github PR can only be used for debugging

thanks

logikal, 2025-05-30:
If there isn't a way to have separate keys for development vs release builds, I guess that makes sense.

Thanks for the attention, and glad we found a workaround!