bambulab/BambuStudio #6993: Unable to select the filament for my A1 with no AMS

luckyp70, 2025-05-21:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.5

### Printer

Bambulab A1

### How to reproduce

After selecting Print Plate, the Send button is disabled. A warning appears on the filament, but it's impossible to select it.

### Actual results

Warning on the only available filament (no AMS), send button disabled.

### Expected results

No warning on the filament, send button enabled.

### Project file & Debug log uploads

It happens for all projects, the issue does not depend on the specific project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

luckyp70, 2025-06-02:
Here is the solution I found on reddit: https://www.reddit.com/r/BambuLab/comments/1kh9deo/error_unable_to_automatically_match_to_suitable/
Thank you for your time and effort spent to solve this issue.

closed 2025-06-02 by luckyp70

XinZhangBBL, 2025-06-03:
@mcorrigan @luckyp70 please try the lateset version.
[image]