bambulab/BambuStudio #6995: Settings window locks out the app when moved to a different monitor.

bull25, 2025-05-21:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.5

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Bambu Studio
2. Click Prepare tab
3. Click either Printer or Project Filaments setting icon
4. Move window to second monitor (Samsung U32R59x)

### Actual results

- Settings window disappears
- App is locked out as you can't click on it because it is expecting input on the window that has disappeared
- Restart of app is needed
- When app is closed in this state, the settings window flashes for a fraction of a second, then the app closes

### Expected results

- App window does not disappear when moving it to second monitor.

### Project file & Debug log uploads

- This happens with no project file loaded
- 

[Bambu Studio - Log Files.zip](https://github.com/user-attachments/files/20375674/Bambu.Studio.-.Log.Files.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-05-22:
Hello, we will conduct an analysis~

Haidiye00, 2025-05-22:
Okay, if there are no more issues, we'll close this order for now~

closed 2025-05-22 by Haidiye00

bull25, 2025-05-24:
Why are we closing this? Can I have the results of the analysis?