bambulab/BambuStudio #6997: AMS problem

TOGCutters, 2025-05-21:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1.  Select "Slice project"
2.  Select "Print Plate"
3.  Select the filament to use
4.  Wait for bed to heat up

### Actual results

 The machine doesn't ever get past 2-9 percent before the project times out and a Warning box pops up saying....

Failed to get AMS mapping table: please select "Resume" to retry (07FF-8012 190708)

The filament stops purging into the disposal area and it just sits with fan on almost like it has stalled.  

I stop the print from the screen on the printer and restart it, selecting the PMC to use directly from the screen and this print seems to work just fine.  It is very inconvenient however as I need to be able to print from my laptop and not have to keep stipping and starting prints.  

this problem began as soon as I updated the firmware.  From that moment on I have not been able to print from my laptop past a 2% mark of the project.  

Please help.....I need this for my business.

[bambu printer (2).zip](https://github.com/user-attachments/files/20377551/bambu.printer.2.zip)
[bambu printer.zip](https://github.com/user-attachments/files/20377550/bambu.printer.zip)

### Expected results

After heating up the bed, purging the old PMC and cleaning off the nozzle.... the project should continue through the "job" and get from 2% to 100% instead of stalling at the 2% mark.

### Project file & Debug log uploads

[debug_Wed_May_21_18_31_54_18500.log.zip](https://github.com/user-attachments/files/20377558/debug_Wed_May_21_18_31_54_18500.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-22:
@TOGCutters Could you please update to the latest version V2.0.3.54 available on the official website and try again? Download link: https://bambulab.cn/zh-cn/download/studio

hwmaier, 2025-05-24:
Had the same problem. After updating to Bambulab V2.0.3.54 problem is fixed.