bambulab/BambuStudio #7002: Bambu Slicer crashes upon clicking preview

hroyalea, 2025-05-23:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 9 5900X 12-Core
Memory: 32GB
GPU: NVIDIA GeForceRTX 3060

### Printer

Bambu A1

### How to reproduce

1.) Open Bambu slicer or .stl file with Bambu Slicer as automatic open setting
2.) Regular edits to slicer file i.e. infill, speed, wall layers, etc.
3.) Click "Preview"

### Actual results

File loads for a moment, freezes briefly, then crashes (automatically closes out)

https://github.com/user-attachments/assets/4873cd09-b515-4c5b-809b-54a3de911a85

### Expected results

Program should show layer slice and print info.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20402410/log.zip)

[blank project file.zip](https://github.com/user-attachments/files/20402431/blank.project.file.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-23:
@hroyalea Hello, I can't open the 3mf file in the attachment above. Could you please provide the original STL file?

QingZhangBambu, 2025-05-23:
@hroyalea please upload a new 3mf file

[image]

hroyalea, 2025-05-23:
@BoBoYu-BamBu @QingZhangBambu @Haidiye00 the 3mf file did not have any data because I provided a blank project file, I was confused by some of the instructions. Here is one of the files I tried using, however they all did not work for me.

[test file debug example.zip](https://github.com/user-attachments/files/20419259/test.file.debug.example.zip)

Haidiye00, 2025-05-24:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657

[image]
Hello, could you please help test this temporary package? Thank you.
We expect to release a new version this month.

Haidiye00, 2025-05-26:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657/artifacts/3189644769
Hello, I have compiled a temporary package here. Can you download it and help me see if it can slice without crashing~thanks

Haidiye00, 2025-06-03:
Hello, please use the latest version of the software. Thank you~
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59

[image]
If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

closed 2025-06-03 by Haidiye00