bambulab/BambuStudio #7003: Can't load 3mf file correctly created by SolidWorks 2022

michelangelo-in-sistine, 2025-05-23:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

3487030@qq.com

### How to reproduce

open it in bamboo studio

### Actual results

[image]

[image]

### Expected results

Microsoft 3d viewer open it correctly

[image]

### Project file & Debug log uploads

[bottom_outer_tube2.zip](https://github.com/user-attachments/files/20402618/bottom_outer_tube2.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-23:
Hello, can you provide your original file? Thank you

MackBambu, 2025-05-25:
@michelangelo-in-sistine 
Do you mean that the STEP file cannot be loaded correctly?
Could you provide the STEP file? Otherwise, it's difficult to know what went wrong during the conversion.