bambulab/BambuStudio #7017: Wont launch due to libOSMesa.so.8 update / dependencies. Garuda linux

Supersudo2000, 2025-05-24:
### Bambu Studio Version

bambustudio-bin-02.00.03.54-1

### Where is the application from?

Built from source by yourself or others

### OS version

Garuda Linux 250308 | 6.14.7-zen2-1-zen

### Additional system information

AMD Ryzen 9 5900X
NVIDIA GeForce RTX 3080
Memory:  31.24 GiB

### Printer

Bambu lab A1

### How to reproduce

Launch Bambustudio from terminal or application directly with mesa being upgraded
pacman -Q | grep mesa
lib32-mesa 1:25.1.1-1
lib32-mesa-demos 9.0.0-3
mesa 1:25.1.1-1
mesa-demos 9.0.0-6
mesa-utils 9.0.0-6

### Actual results

$ bambu-studio 
/opt/bambustudio-bin/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory
Downgrading mesa works, but program stutters like crazy

### Expected results

program launches

### Project file & Debug log uploads

 bambu-studio 
/opt/bambustudio-bin/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory

### Checklist of files to include

- [ ] Log file
- [ ] Project file

madmalkav, 2025-05-24:
Already covered by issue 6951.

c2h5oh, 2025-05-25:
> Please update the libOSMesa library

It's a part of Mesa, not BambuStudio. Latest Mesa release (25.1) has removed it completely after it has been deprecated for quite a while.

ShiroBlank, 2025-05-26:
I had success temporarily moving over to the flatpak version + nvidia workaround until this is resolved.

notkaramel, 2025-05-27:
Got the same problem on Arch Linux after updating all packages. I downgraded `mesa` to version 1:**24.3.4-1** and BambuStudio seems to work again.

Downgrade program: https://aur.archlinux.org/packages/downgrade

```sh
sudo downgrade mesa
```

notkaramel, 2025-06-09:
> Don't understand why you are being downvoted. Your comment exactly solved the issue, thank you.

It occurs to me that `mesa` versions before 25 could break other stuffs, hence why my solution was not a truly great idea. Use `downgrade` with caution, I guess :)

Mesa version 24 broke `hyprland` and `alacritty` for me today after a system update (unsure how to traceback all the packages I updated).

thoastbrot, 2025-06-11:
https://github.com/bambulab/BambuStudio/issues/7213 recommends https://aur.archlinux.org/packages/bambustudio-nvidia-bin - works fine for me without other tweaks.

Beagon, 2025-07-02:
This issue is also encountered on Ubuntu 25.04

LordyPandaz, 2025-07-06:
Mesa 25.1 intentionally removes libOSMesa, which every linux distribution is using.  This needs to be updated, downgrading Mesa isn't a fix.

MackBambu, 2025-07-28:
Please use the Flatpak version temporarily until the issue is resolved.