bambulab/BambuStudio #7019: flow dynamics auto calibratiojn fail for external spool

imdocwho, 2025-05-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 Pro version 24H2 OS build 2610.4151

### Printer

Bambulabs X1C

### How to reproduce

click calibration > Flow dynamics > auto-calibration
sync ams / nozzle info
Sync filament calibration
click external spool
click calibrate

### Actual results

invalid tray_id or use_ams flag

[image]

### Expected results

to run the test

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

changbaishen, 2025-05-27:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

vanowm, 2025-06-01:
@changbaishen this is about BambuStudio, not a phone app...and it's reproducible with LAN mode

People suggest this is actually the new firmware issue.