bambulab/BambuStudio #7020: Can't utilize full right nozzle only print area (

Mah-AES, 2025-05-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU:  Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz   2.11 GHz
RAM:  16 GB

### Printer

Bambu Lab H2D

### How to reproduce

In attempting to print the attached duct file, my object falls within the published printable area for the right nozzle only, but will not slice and the error "Objects are laid over the bounds" pops up.  

I'm basing the printable area on this offical Bambu Lab Wiki article:  [image]  

Project file is submitted below.

### Actual results

This is the size, positioning of my object, and error pop up received in Bambu Studio:  

[image]

### Expected results

Based on documentation for the printer, this should be within the printable area for the right nozzle.  I know this technically isn't a "bug" but the software is not allowing the full stated capability of the printer.

### Project file & Debug log uploads

[Diffuser - Diff Sect 1.zip](https://github.com/user-attachments/files/20428125/Diffuser.-.Diff.Sect.1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-05-26:
@Mah-AES  Because the position of the object in your file does exceed the printable area of ​​the plate.
I fine-tune the coordinate position of the model and uploaded the modified model file.

[Diffuser - Diff Sect 1-modified.zip](https://github.com/user-attachments/files/20440898/Diffuser.-.Diff.Sect.1-modified.zip)

zhimin-zeng-bambulab, 2025-06-26:
It's related to  coordinate accuracy, not a bug.