bambulab/BambuStudio #7021: Prime tower moves after chaning any slicer settings

mzient, 2025-05-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

BambuLab P1S

### How to reproduce

1. Slice a multi-material plate
2. Move the tower to a tight spot between the object and plate edge (see image)

[image]

3. Change anything in slicer settings.

[image]

5. Re-slice

### Actual results

The tower moves and overlaps with the object.

[image]

### Expected results

The tower stays exactly where it was before.

### Project file & Debug log uploads

[wipe_tower_bug_min_repro.zip](https://github.com/user-attachments/files/20428211/wipe_tower_bug_min_repro.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-05-26:
Thanks for reporting this issue, we will fix it.

zhimin-zeng-bambulab, 2025-06-05:
The new version ([v02.01.00.59](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59)) has been optimized for this problem, you can have a try.

closed 2025-06-17 by zhimin-zeng-bambulab