bambulab/BambuStudio #7022: Can't login. Login window is empty.

LurkusMurkus, 2025-05-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro. 24H2. Build: 26100.4061

### Printer

Bambu Lab A1

### How to reproduce

Start Bambu Studio, press login.
Or try sending anything to the printer.

### Actual results

Empty Login window pops up. Nothing to press, no other way to login.

### Expected results

Expected to login and send files to the printer.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

2coldz, 2025-05-25:
Have exact same problem, can't login just blank page.

BoBoYu-BamBu, 2025-05-26:
@2coldz Thank you for your feedback. We are aware of this issue. Please switch to another language, then switch back and try again.