bambulab/BambuStudio #7028: Crash when slicing large file: Program closes without error message

MK141998, 2025-05-26:
**Thank you for using Bambu Studio and wanting to report a bug.**

---

**Bambu Studio Version**
2.0.3.54

**Where is the application from?**
Bambu Lab Official website

**OS version**
Windows 11

**Additional system information**
_No additional system info provided._

**Printer**
Bambulab H2D

**How to reproduce**
1. Load a large file in Bambu Studio
2. Attempt to slice the file
3. The program terminates without an error message and closes automatically
4. Reopening the program and reloading the same project, slicing again causes the same crash

**Actual results**
The program closes automatically without any error message whenever slicing a large file. Upon reopening, the project can be loaded again, but slicing results in another crash.

**Expected results**
The program should successfully slice large files without crashing or at least provide an error message if it cannot.

**Project file & Debug log uploads**
_User cannot provide the project file. Debug log file is attached below._

**Checklist of files to include**
- [x] Log file (attached)
- [ ] Project file (not available)

JohnSmithCYM, 2025-05-27:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657/artifacts/3189644769
@Haidiye00  have compiled a temporary package here. Can you download it and help me see if it can slice without crashing~thanks

Haidiye00, 2025-06-04:
Hello, please use the latest version of the software. Thank you~
[Release 2.1.0 Public Release · bambulab/BambuStudio](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59)

[image]
If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software