bambulab/BambuStudio #7035: 自适应立方体（以及别的图案）填充不能识别内部悬空墙

SODIUMWATER, 2025-05-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

WIN11 24H2

### Additional system information

XPS 15 9520
12900HK 32GB
3050TI

### Printer

X1CARBON/H2D

### How to reproduce

切片

### Actual results

两种材料交界处内外墙浮空（悬垂）

[image]

### Expected results

为内外墙下面添加必要的桥接和实心填充或更改内外墙形状以适应稀疏填充图案

### Project file & Debug log uploads

[POLY_3COLOR_H2D.zip](https://github.com/user-attachments/files/20453921/POLY_3COLOR_H2D.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

hotellonely, 2025-05-30:
自适应填充的bug非常之多，建议直接放弃

QingZhangBambu, 2025-06-03:
@XunZhangBambu, please check this.