bambulab/BambuStudio #7045: [BUG] H2D missing image and filename being printed is replaced by path now

bolsoncerrado, 2025-05-29:
### Bambu Studio Version

Latest Beta BambuConnect available

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma

### Additional system information

Chrome browser....

### Printer

H2D Pro

### How to reproduce

Launch a print from BambuStudio....

### Actual results

It's missing the H2D pic.
And now, instead of the filename being printed, it shows the PATH instead:

[image]

### Expected results

H2D pic being displayed.
Filename/Platename displayed instead.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-29:
@bolsoncerrado Thank you for your feedback.Could you please zip up the entire log folder shown in the picture and provide it to us?
[image]

bolsoncerrado, 2025-05-29:
But this is related to BambuConnect. Not to BambuStudio. :???

BoBoYu-BamBu, 2025-05-29:
> [@bolsoncerrado](https://github.com/bolsoncerrado)感谢您的反馈。您能否将图中所示的整个日志文件夹压缩并提供给我们？ [image]

Could you please first provide the log for us to analyze? We still need to investigate the specific cause.