bambulab/BambuStudio #7049: Preview for Print time columns are not displayed correctly (not readable)

DevTKSS, 2025-05-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

Graphic card Geforce RTX 4070 TI
128gb ram
intel 13th gen CPU

### Printer

X1C

### How to reproduce

slice a long running print with multiple plates

### Actual results

for reference, this showed up at the MakerWorld uploaded Project, downloaded to Bambu Studio with applying Bambu Matte PLA original Profiles instead of the preset generic ones:

[image]

### Expected results

the table alignment should align correctly, so the right column would be acutally readable ;)

### Project file & Debug log uploads

https://makerworld.com/de/models/796921-helldivers-2-ar-23-liberator-replica X1C Profile with above mentioned changes

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-06-03:
Hello, we will fix it~

Haidiye00, 2025-07-18:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

closed 2025-07-18 by Haidiye00