bambulab/BambuStudio #7053: Bambu Studio showing different print colors based on bambu instance

Eragonsaphria, 2025-05-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Not relevant

### Printer

X1c + Combo

### How to reproduce

I printed from printer itself 
Have 2 instances of Bambu Studio 
Have one color selected as main color in one instance, have a differenct color in the other. 
The on going print will show 2 different colors based on the selected filament.

### Actual results

[image] - location to look at

[image]- filaments from instance 1 

[image]- filaments from instance 2

### Expected results

You should see 2 identical windows with the color being different based on the instance running.

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [ ] Project file

Eragonsaphria, 2025-06-04:
Thanks for the response, I will try and replicate it myself now that my printer queue is done 

 If you want to try and replicate it do as I said, 
Have 2 instances of the slicer open, have sliced models in each, have each one with a single filament varying in color. Use the Printer to start the print from print files stored locally on the printer via sd card. Start Print. 

Go back to computer and check if the task shows different colors on the task in operation menu on the bottom of the screen.

Eragonsaphria, 2025-06-04:
[image]

neither of these are updating but this is where i would expect to see different colors

Haidiye00, 2025-06-05:
[image]
Hello, I haven't fully understood your request yet. I'm very sorry, could you please provide a more detailed description of your problem, combined with pictures or GIFs.
Additionally, if you open multiple Bambustudio files, they will all access and modify this configuration file simultaneously. I think it is highly likely that there may be conflicts in expecting different consumables for each Bambustudio.

Eragonsaphria, 2025-06-05:
"Additionally, if you open multiple Bambustudio files, they will all access and modify this configuration file simultaneously. I think it is highly likely that there may be conflicts in expecting different consumables for each Bambustudio."

This may be the cause. Like I said I started a print from the X1c and the slicer appeared in the wrong color on this icon 

[image]

I expected it to be white and it was yellow on one instance. So maybe when starting a print from the X1c it will use the .conf file to guess which color to make this image incorrect 

[image]

It does not occur if I start the print from the PC itself only if I start from the printer.