bambulab/BambuStudio #7062: Adjust fans for X1C in Bambu Studio

Esofasto, 2025-06-01:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 / 11

### Additional system information

Not required - software bug

### Printer

X1C

### How to reproduce

Go to device screen
Click on Fan to try and adjust fan speed

### Actual results

Unable to change fan speed settings

### Expected results

Should be able to adjust fan speed(s)

### Project file & Debug log uploads

Anyproject.zip

### Checklist of files to include

- [x] Log file
- [x] Project file

Esofasto, 2025-06-01:
No other comment to make.

gallowayk, 2025-06-01:
Same issue here.

changbaishen, 2025-06-04:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

sonjamichelle, 2025-06-07:
[image]

Clicking on fans only shows you a description of what the fans do instead of allowing you to control the fans as in previous versions.