bambulab/BambuStudio #7065: Selecting text font nearly impossible

gr3333, 2025-06-01:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5 (24F74)

### Additional system information

Apple M3 (MacBook Air)
8 GB RAM
Built in Graphics

When I try to select a font for adding text to a model, the selection window is fairly small and when I try to scroll, it jumps all over the list and there's no way to scroll down the list.

[image]

### Printer

Bambu P1-s

### How to reproduce

got to the text option
try to scroll down the list 

### Actual results

It jumps / scrolls way too fast

If there was a scroll bar, or the arrow keys could be used to move down the list it would work

a bigger/longer list might help

### Expected results

a controlled way to navigate the list and select the desired font.

or a search feature to filter the list

### Project file & Debug log uploads

this is not dependant on any file, it happens in all situations.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-03:
[image]
Hello, there is a slider here

eried, 2025-06-03:
I guess nobody struggles due using the mouse scroll wheel or touchscreen (which messes up bambustudio in Windows)... but maybe the scrollbars should be unified?

[image]

Haidiye00, 2025-06-03:
Hello, you are right,we are considering increasing the width of the slider a bit

VimanyuAgg, 2025-06-21:
It will be much better if we can also scroll using arrow keys on the keyboard. I'm not able to select font one by one. I'm on macOS.

Haidiye00, 2025-07-18:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

closed 2025-07-18 by Haidiye00