bambulab/BambuStudio #7070: Error “invalid tray_id or use_ams flag” when using auto calibration

carlpatt, 2025-06-02:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

PopOS 22.04

### Additional system information

FW 01.09.00.01 causing issues with auto calibration for external spool users. [Multiple users reported this on a community thread](https://forum.bambulab.com/t/error-invalid-tray-id-or-use-ams-flag-with-last-firmware/165675). 

### Printer

X1C

### How to reproduce

1. upgrade x1c fw to 01.09.00.01
2. go to calibration window in bambu studio
3. click on auto calibration
4. check box for external spool
5. click on calibrate

### Actual results

error message “invalid tray_id or use_ams flag” pops up. 

### Expected results

The auto calibration feature should print  -> scan -> show results. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

carlpatt, 2025-06-05:
@changbaishen my ticket number is US250605166001

Bax85, 2025-06-12:
Trying to do a **flow rate calibration** for **External** Bambu TPU 95 HF.
Getting same error **"invalid tray_id or use_ams flag"** when starting calibration.

Seems like Flow dynamics Calibration is working.

Bambu Lab X1 Carbon
V.01.09.01.00

AMS-1
V.00.00.06.75