bambulab/BambuStudio #7072: Bambu studio software crash when opening Variable Layer Height

Klan33, 2025-06-02:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

Windows surface ARM
Prozessor	Snapdragon(R) X 12-core X1E80100 @ 3.40 GHz   3.42 GHz
Installierter RAM	16.0 GB (15.6 GB verwendbar)
Systemtyp	64-Bit-Betriebssystem, ARM-basierter Prozessor
Stift- und Toucheingabe	Unterstützung der Stift- und Toucheingabe mit 10 Touchpunkten

### Printer

H2D

### How to reproduce

open empty file
import a model
left click on model to activate
just click on icon Variable Layer Height
crash
tried with several easy models

### Actual results

crash of of bambu studio

### Expected results

feature works well

### Project file & Debug log uploads

[crash_Mon_Jun_02_13_35_21_0_done.log](https://github.com/user-attachments/files/20550597/crash_Mon_Jun_02_13_35_21_0_done.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-03:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)
And can you provide your 3mf file?

Lawdawgy, 2025-06-04:
I have the same issue

Klan33, 2025-06-06:
[image]
I uploaded a gif. But I'm not able to upload a 3mf, because this system does not allows to upload such format. 
I also tried to import different CAD file types like STL, STEP AP203 and 3mf. Anytime the same: when I click on the icon VARIABLE LAYER HIGHT a crash appears.

Klan33, 2025-06-06:
[Bug files.zip](https://github.com/user-attachments/files/20626438/Bug.files.zip)
here they are

lanewei120, 2025-06-10:
Hello @Klan33 

could you also help to upload this file OpenGlOn12.dll here?
we can have more check for the stacks

as we can not build a runnable version for windows arm currently(the released version is windows x64 which can also run on windows arm)
we are hard to debug the issues on windows arm platform