bambulab/BambuStudio #7074: [PL] AMS settings button missing with Polish language

kajak4u, 2025-06-02:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Have P1S printer with AMS
2. Have Bambu Studio with language set to Polish
3. Go to Device page and connect to the printer (LAN mode)

### Actual results

AMS settings button is not present

[image]

### Expected results

AMS settings button should be visible, as per other languages (English below)

[image]

### Project file & Debug log uploads

(not related to any project)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-03:
@kajak4u Thank you for your feedback. The internal version of this issue has been fixed. Please wait for the new version.

BoBoYu-BamBu, 2025-06-03:
@kajak4u Our new version has been released. Please update and verify this issue. Thank you.

closed 2025-06-03 by kajak4u