bambulab/BambuStudio #7085: Tree support does not always support itself

ithinu, 2025-06-03:
### Bambu Studio Version

2.1.0

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 24

### Additional system information

I attach a screenshot and the project file.

[image]

### Printer

Bambu Lab A1

### How to reproduce

Open the attached project. It has enabled Manual Tree support. The screenshot shows the problem.

### Actual results

Arcs few mm outside support. Likely a support point close to the trunk made it shift like that.

### Expected results

A support tree that supports itself.

### Project file & Debug log uploads

[tree_bug.zip](https://github.com/user-attachments/files/20572571/tree_bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

hotellonely, 2025-06-03:
https://github.com/bambulab/BambuStudio/issues/7090

I think our issues might be related.

hotellonely, 2025-06-05:
> [@ithinu](https://github.com/ithinu) Thanks for your feedback. This is a known BUG with OrganicTree. The over-smoothing of the branches causes misalignment between upper and lower layers and we are fixing it. For now, I recommend trying HybridTree or other types of supports.

However for other tree types, it would grow through the part or grow right on top of the top surface of the part, regardless of whether "only on plate" is selected, as I have described in my issue report.

ithinu, 2025-06-05:
> [For now, I recommend trying HybridTree or other types of supports.

@NothFie, the problem was that all the other types of tree supports were leaning in a way that made me afraid about whether the model would be rigid enough to ensure good layer adhesion. See the screenshot of the hybrid trees below, along with a sketch showing how the tree could be made much more rigid using the same amount of support material.

[image]

hotellonely, 2025-06-06:
> [@NothFie](https://github.com/NothFie), the problem was that all the other types of tree supports were leaning in a way that made me afraid about whether the model would be rigid enough to ensure [...]

This can be adjusted by changing the branching angles.

NothFie, 2025-06-06:
@ithinu I recommend you disable the option "independent support layer height" and set the wall count to 2. I didn't see Brim areas in your picture but they exist when I reproduce the file, I suggest not disabling it. To be safe, you could also increase the "tree_support_branch_diameter_angle" a little(to 7 or 8). I believe these operations are enough to help you print it.

ithinu, 2025-06-06:
@NothFie disabling "independent support layer height" helped a lot and thanks for reminding me about the brimming.
@hotellonely I have those exact settings but "independent support layer height" got in the way.