bambulab/BambuStudio #7088: Umrandung der Stützstruktur geht durch die Wände des Objektes.

Atzi92, 2025-06-03:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu H2D

### How to reproduce

1. Aktivieren sie die Umrandung eines Teils welches eine Stützstruktur im Teil besitzt.

[image]

### Actual results

[image]

### Expected results

Es dürfte keine Umrandung durch das Bauteil gehen.

### Project file & Debug log uploads

[Comfee Schlauch Trichter v1 Brim Fehler.zip](https://github.com/user-attachments/files/20573801/Comfee.Schlauch.Trichter.v1.Brim.Fehler.zip)

[log_iotc.txt](https://github.com/user-attachments/files/20573834/log_iotc.txt)

### Checklist of files to include

- [x] Log file
- [x] Project file

NothFie, 2025-06-05:
@Atzi92 Thanks for your feedback. Sorry for your bad experience, we are fixing this BUG and will release it in the next version..

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)