bambulab/BambuStudio #7090: Tree Support Growing Through Model (with Support Interface Filament)

hotellonely, 2025-06-03:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab github releases

### OS version

macOS latest

### Additional system information

M3 Max 64GB+2TB

### Printer

H2D

### How to reproduce

1. Create a new project.
2. Add a Benchy primitive.
3. Enable tree support.
4. Assign a support interface filament (e.g., Support for PLA)

Voila, devilish tree!

### Actual results

[image]

You get a devilish tree that grows through the benchy and ruins the surface quality.

### Expected results

If you do not enable the support filament, the slicing would be correct.

### Project file & Debug log uploads

Simply do what I said and you would see it from happening.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hotellonely, 2025-06-03:
The worst part is that the generated tree not only just goes through the model, it also does not actually attach on the model:

[image]
Upon closer inspection you would notice that the tree is disconnected with everything, and it's completely printing in air.

This is a doomed spaghetti.

hotellonely, 2025-06-03:
@Haidiye00 @QingZhangBambu @XinZhangBBL @zhimin-zeng-bambulab 

This is a quite high impact bug, can we get a hotfix?

Interestingly **Tree(auto), Tree slim, strong, hybrid** all don't work. But Tree Organic still works.

NothFie, 2025-06-05:
@hotellonely Thanks for your feedback. We are urgently fixing the issue.