bambulab/BambuStudio #7094: The latest update broke my filament profile

BogdanMironyuk, 2025-06-03:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I5-10300, RXT3060, 32GB RAM

### Printer

BambuLab P1s

### How to reproduce

1. Go to "set filament to use"
2. Click on custom filament
3. For my case click on "edit" for custim filament
4. Custom filament have not any filament preset (but before last update it was), try to delete this profile

### Actual results

After trying delete this profile not deleted, also i can`t pick this profile in PA settings (but before last update everything worked).

### Expected results

custom profile didn`t brake after update or i can delete this profile.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20577671/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-04:
@BogdanMironyuk Sorry, could you provide us with a screen recording or a screenshot?

BogdanMironyuk, 2025-06-04:
> [@BogdanMironyuk](https://github.com/BogdanMironyuk) Sorry, could you provide us with a screen recording or a screenshot?

Hello. I`ve made a video and download on youtube: https://www.youtube.com/watch?v=SJFoIzWN-jo

digitalresistor, 2025-06-05:
@BogdanMironyuk I am having the same issues, when I go to Custom Filaments and click edit, it shows no filament profiles.

Also see this issue on Reddit: https://www.reddit.com/r/BambuLab/comments/1kw0qha/problems_with_filament_profiles/ with the same problem.

@BoBoYu-BamBu something is corrupting the list of filaments in BambuStudio/overwriting the values when that filament is loaded in the AMS. This is causing a lot of pain as I keep having to re-create the filament profile and add it back to the printer in Bambu Studio.

BogdanMironyuk, 2025-06-14:
> [@BogdanMironyuk](https://github.com/BogdanMironyuk) Sorry, could you provide us with a screen recording or a screenshot?

Hi @BoBoYu-BamBu . What about this problem? Do you have any solution? Because i still can`t fix it :(

komp0t, 2025-06-17:
Please fix the problem. I also found this. The profile cannot be deleted, and it disappears from AMS after a couple of minutes after selection.

BogdanMironyuk, 2025-07-26:
@DanBao-Bambu @BoBoYu-BamBu 
Hello! Do you have solution for this issue? On last version 2.2.0.60 I can`t delete or use my filament profile yet. This causes a lot of inconvenienc