bambulab/BambuStudio #7099: Clicking makers logo on profile does not open a page but it change cursor to hand so its detected as a clickable object but nothing happens

trich101, 2025-06-03:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 Pro

### Printer

AI Mini

### How to reproduce

Go to any model
Move cursor to top left corner of popup with makers avatar/logo. You see cursor change from arrow to hand like its selectable
Click
Nothing opens. I assume this is supposed to open makers page on Web to view their collection. Since we cannot search by vendors name, or not nearly as well, in Studio as we can online, I am assuming allowing us to click this logo was a solution. The cursor reacts to it so it makes sense its a link but the popup does not appear.

### Actual results

No page opens when clicking vendor logo in model page

### Expected results

Cursor reacts, turns into hand, like is a link but nothing opens. I would have expected popup to vendors web page or maybe in Studio but that makers full collection.

### Project file & Debug log uploads

No project file. Issue occurs when browsing for any model in Studio.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-04:
@trich101 The issue has been reported to the mall team. They are currently making the necessary corrections. Please look forward to the next mall update. This issue is now closed. Thank you.

closed 2025-06-04 by BoBoYu-BamBu

trich101, 2025-06-06:
This was not on the website, but in the Windows 10 desktop application.

trich101, 2025-06-06:
It is not a hyperlink issue on the html website, but the standalone studio application. I filed this from within the application and since I work OPs in IT as well and have seen thousand tickets in the last 10 years, this feels like it was read as reporting a website error and it was sent to web support. I just want to make sure its understood this was the desktop app, NOT the website. I would assume the open support from the Studio app is used for Studio support.