bambulab/BambuStudio #7102: starting new pring

g-burch, 2025-06-03:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

when clinking send to printer produces a blank/transparent window.  Same also happens when trying to print file from SD card.

### Actual results

[image]

### Expected results

print my file

### Project file & Debug log uploads

[debug_Tue_Jun_03_13_16_35_15064.log.zip](https://github.com/user-attachments/files/20580306/debug_Tue_Jun_03_13_16_35_15064.log.zip)

[NEW STAND OFF.zip](https://github.com/user-attachments/files/20580309/NEW.STAND.OFF.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-06-26:
Thank you for your feedback. what's your OS version in detail.

XinZhangBambu, 2025-08-27:
Thank you for your feedback. We have developed a software version to address your issue; could you please verify if it works? If it does, we will merge it into the release version.
https://github.com/XinZhangBambu/BambuStudio_bug_fix/actions/runs/17255536799#artifacts

[image]