bambulab/BambuStudio #7108: Wall Slicing problem

Protey88, 2025-06-04:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10

### Printer

Babmu LAB A1 Combo

### How to reproduce

Press slice plate! And see on model (Lid Plate)

### Actual results

Wall slicing artefact on lid model

### Expected results

Normal wall slice without artefacts

### Project file & Debug log uploads

[LID MODEL Problem REPORT.zip](https://github.com/user-attachments/files/20585751/LID.MODEL.Problem.REPORT.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-BamBu, 2025-06-04:
@Protey88  Thank you for your feedback. Could you please provide a screenshot?

Protey88, 2025-06-04:
[image]

[image]

Haidiye00, 2025-06-04:
[image]
I turned off this preference, and the actual slicing result is like this