bambulab/BambuStudio #7119: Uncaught exception: ConfigOptionVector::set_only_diff(): Assigning from an vector with invalid diff_index size

briandaymsft, 2025-06-04:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: i7-1265U
Mem: 32GB
Display: Intel Iris Xe Graphics (2.0 GB)

### Printer

Bambu Lab H2D

### How to reproduce

1. Uninstall Bambu Studio, and remove user data.
2. Install Bambu Studio
3. Choose North America
4. Choose H2D printer
5. Answer yes to if you want to sync your Bambu Lab cloud data locally.

### Actual results

Unhandled unknown exception; terminating the application.

This earlier today. I had been using the software without problems, then upon reopening it was faced with this unhandled exception. The only way to get around the error is to uninstall and remove user data. Doing this lets you reinstall and launch the app. However, as soon as I try to sync my cloud data locally the exception immediately pops and crashes the app.

The log files have the same error each time just after updating filament compatibility.

```
2025-06-04 15:02:11.226148[Thread 0x00002b14]:Uncaught exception: ConfigOptionVector::set_only_diff(): Assigning from an vector with invalid diff_index size
2025-06-04 15:02:11.226148[Thread 0x00002b14]:3:02:11 PM: Error: BambuStudio got an unhandled exception: ConfigOptionVector::set_only_diff(): Assigning from an vector with invalid diff_index size
2
```

[image]

[debug_Wed_Jun_04_15_51_53_6104.log](https://github.com/user-attachments/files/20598044/debug_Wed_Jun_04_15_51_53_6104.log)

### Expected results

Application continues to function.

### Project file & Debug log uploads

The issue can be reproduced 100% on my machine with no project file open after a clean install of Bambu Studio when trying to sync cloud settings locally.

### Checklist of files to include

- [x] Log file
- [ ] Project file

briandaymsft, 2025-06-04:
A temporary workaround appears to be disabling the preference "Auto sync user presets (Printer/Filament/Process)"
When the unhandled exception box appears, you can still click the main Bambu Studio window and get to the Preferences menu. Disabling the above option, then restarting, results in no more unhandled exception error. If you enable the option again, the error immediately returns.

lanewei120, 2025-06-05:
hello @briandaymsft 

it seems your user presets corrupt
from log, it is very likely caused by this one
Bambu PLA Translucent @BBL H2D - Copy

could you help to upload this preset json file from directory

[image]

thanks

briandaymsft, 2025-06-05:
Hi @lanewei120 , all of my user\ID\* directories are currently empty due to a previous uninstall where I chose to remove user data. There are no files from the current installation to share from this location. Is the problem sync'ing the file(s) from the cloud side back to my machine?

I did create a backup of my user directory before I uninstalled Bambu Studio and let it delete user data. The file **Bambu PLA Translucent @bbl H2D - Copy.json** you requested does not exist in the backup. There is a **Bambu PLA Translucent @bbl H2D - Copy.info** file, but no .json equivalent.

Just to repeat, the actual user\ID\filament directory the installation is using is empty at this time.

lanewei120, 2025-06-06:
> Hi [@lanewei120](https://github.com/lanewei120) , all of my user\ID* directories are currently empty due to a previous uninstall where I chose to remove user data. There are no files from the [...]

Hello @briandaymsft 
thanks for this information
we just found some mistake with "Bambu PLA Translucent", and will fix it soon

Gerard-Martinez, 2025-06-09:
When I open Bambu Studio, I get this error and it won't let me do anything. I don't know how to fix it.
If I reinstall the program, my printer and printing configuration data will be deleted.
When reinstall bambu studio and uploading my configuration from Bambu Cloud, I also get the error.
A temporary workaround appears to be disabling the preference **"Auto sync user presets (Printer/Filament/Process)".**
If I open the program without Wi-Fi, I don't get the error. I'm sure it's trying to open a faulty MakerWorld project or something.
I attached a screenshot of the task manager.

Windows 10
Bambu Studio 2.1.0.59 (latest version on 06-09-2025)
Region : EU
Model : X1 Series
State/Province: Balearic Islands
Country/Region: Spain

[image]

[image]

[crash_Mon_Jun_09_18_35_02_0_done.log](https://github.com/user-attachments/files/20658466/crash_Mon_Jun_09_18_35_02_0_done.log)

[log_iotc.txt](https://github.com/user-attachments/files/20658469/log_iotc.txt)

Please tell me what file you need to see the error and fix it. 

Thank you very much!

lanewei120, 2025-06-10:
Hi @Gerard-Martinez 

we will fix this issue soon
Currently could you just disable the option "Auto sync user presets (Printer/Filament/Process)"?

briandaymsft, 2025-06-17:
> Hi [@Gerard-Martinez](https://github.com/Gerard-Martinez)
> 
> we will fix this issue soon Currently could you just disable the option "Auto sync user presets (Printer/Filament/Process)"?

Hi @lanewei120 ,

I upgraded to 2.1.1.5.2 after seeing the below in the release notes, and my unhandled expected exception no longer repros with the sync option enabled once again. 

_**Fixed the issue where third-party preset synchronization could cause crashes, and also fixed errors in parsing and displaying certain parameters. The corresponding flush temperature parameter option has been temporarily disabled.**_

Was the fix you expected deployed in this build? Thank you.

briandaymsft, 2025-06-18:
Thank you very much for the confirmation and quick fix, @lanewei120 .

closed 2025-06-18 by briandaymsft

Gerard-Martinez, 2025-06-18:
I enabled Auto Sync User Presets option (Printer/Filament/Process) with version 2.1.1.5.2, and the error no longer appears. Thanks so much for the fix!!!