bambulab/BambuStudio #7121: Violation d'accès

filoutom, 2025-06-04:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1 Mini

### How to reproduce

Lors de Trancher le plateau

### Actual results

a Fatal Error occured: "Violation `d'accès"

[image]

### Expected results

pas d'erreur

### Project file & Debug log uploads

[debug_Wed_Jun_04_21_44_41_21252.log.zip](https://github.com/user-attachments/files/20598303/debug_Wed_Jun_04_21_44_41_21252.log.zip)

[A1-P1-X1 Combi 4x4.zip](https://github.com/user-attachments/files/20598313/A1-P1-X1.Combi.4x4.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-06-06:
@filoutom  Thanks for reporting this issue, we will fix it in the next version, you can avoid this problem by first using  By-layer printing mode.

zhimin-zeng-bambulab, 2025-06-16:
It has been fixed in v02.01.01.52, you can have a try.

closed 2025-06-16 by zhimin-zeng-bambulab