bambulab/BambuStudio #7124: Latest update of Bambu Studio does not allow sign in

Halterra, 2025-06-05:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 10.0.26100 Build 26100

### Printer

Bambu Labs P1S

### How to reproduce

I open Bambu Studio and the following error occurs

### Actual results

[image]

### Expected results

There are no preferences or printers loaded

### Project file & Debug log uploads

No files available.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-06-05:
hello @Halterra 

could you delete this directory C:\Users\Robert\AppData\Roaming\BambuStudio\system
and reboot bambu studio for a try

thanks

Halterra, 2025-06-05:
Couldn’t find the directory.  Sincerely,Robert Haller

lanewei120, 2025-06-05:
how about enter the folder from here

[image]