bambulab/BambuStudio #7131: Color Scheme now defaults to FILAMENT (used to be LINE TYPE) -> PUT THIS IN SETTINGS?!?!

bolsoncerrado, 2025-06-05:
### Bambu Studio Version

2.1.0.5.9

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Somona

### Additional system information

M1Ultra

### Printer

N/A

### How to reproduce

Slice any part

### Actual results

Whatever I slice, the default color scheme is anything but LINE STYLE like it used to be.

### Expected results

Default to LINE STYLE *OR* make it user selectable on settings menu

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-06-06:
@bolsoncerrado  Currently, the default parameter displayed after slicing is "Filament". However, you can click to select the parameters you want to view. Please refer to the attached image for details.

[image]

bolsoncerrado, 2025-06-06:
I know I can click on that at any time. But it used to be DEFAULTED TO LINE TYPE and I'm *VERY* used to that view, since the early Simplify3D days, so maybe a USER SETTING to choose which COLOR SCHEME to use as default would be great.

BoBoYu-BamBu, 2025-06-06:
OK, we need to make an assessment.

theaecologist, 2025-06-08:
I've been battling with this all day. I think it was the most recent Studio update that changed this behaviour.
I'm happy if it simply remembered the last used setting.

Brausmith, 2025-06-08:
I've been battling with this all day. I think it was the most recent Studio update that changed this behaviour.
I'm happy if it simply remembered the last used setting.

bolsoncerrado, 2025-06-08:
Last used would work for me as well if that's easier to implement, definetly!!

Serendipisea, 2025-06-14:
Any news on the assessment?  This is so tedious to change every single time I print something.  I'm going to try out another slicer.

RandJo3, 2025-06-19:
struggling with this as well, pretty darn annoying!! It was pretty jarring to see a solid colored block when I sliced an object. Sucks to have to click through extra menus, I just want to easily discern where my support starts and my print begins! I thought that I would be able to change the default view in the Preferences menu (ctrl+P) but the option isn't there.

JesseCrowleyPhoto, 2025-06-23:
Please please please make this a setting that users can choose the default, or just got back to line type as default. Filament is almost useless. 

Please and thank you.

SiJiL82, 2025-07-01:
Filament as the default is basically useless - when everything is just a big solid colour it's impossible to see where things like trees have applied, or where overhangs are that you need to put a tree on. Definitely needs to be a configurable option.

wildwoodashes, 2025-07-14:
Agreed. This is a huge oversight to not allow the user set the default. As everyone else has said, the 'filament' line type is nearly useless. It's fine if that's the out of the box (even though it's useless), but not giving us an option to set our own default is super annoying.

Brausmith, 2025-07-14:
> OK, we need to make an assessment.

Why is this taking so long?
Regardless of the decision, the assessment should have been done by now, with the outcome posted here.

JohnSmithCYM, 2025-07-15:
Thanks for your feedback, we have put it in demand pool and we will implement this soon.

Neto, 2025-07-28:
+1 to implement this. just save last "Slicing Result Color Scheme" and its done.
Thank you!

JohnSmithCYM, 2025-07-30:
We will add a Preference option for users to remember the last time color scheme (before close Studio) then use it in next time startup of Studio's slice.

alexmarles, 2025-08-01:
The proposed solution **does not address the issue described** and veers off industry standards. Consider reopening the issue and make use of the **free** user insights that you are getting here.

bolsoncerrado, 2025-08-01:
> The proposed solution **does not address the issue described** and veers off industry standards. Consider reopening the issue and make use of the **free** user insights that you are getting here.

**BOOM** Micdrop!

Ben parit!! xD

slikts, 2025-11-17:
I do have multi-filament prints often, but defaulting to the filament color scheme by default is still a continuous annoyance, because the line type info is almost always more pertinent. This problem is exactly in the category where you can be sure that it's not important enough for most people to look it up or give feedback, but it is annoying and just being tolerated, until like in my case the annoyance has accumulated enough to spend time looking for a fix.