bambulab/BambuStudio #7139: Camera remains in initializing state after update to firmware  1.8.2.0

FvdM-GH, 2025-06-05:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

W10 64-bit

### Printer

Bambu P1S with AMS combo

### How to reproduce

Occasionally... No specific reason or way to reproduce.

### Actual results

Camera stays in initializing state forever, no visual.

### Expected results

Normal start of the camera, from initializing to loading.

### Project file & Debug log uploads

No log files, program doesn't crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

matt-in-brissy, 2025-06-07:
Yep, I'm having the same issue after updating the printer to the latest firmware too.  I restarted the machine and it worked again for a little while, but then it stopped working again a bit later (after one or two prints).

FvdM-GH, 2025-07-23:
> Hello, could you please reproduce the problem again, then close the software and send a new log? The relevant information was not found in the previous log.

Hi,

Didn’t see it anymore, sorry. Perhaps a firmware glitch?
Regards,

Frank.