bambulab/BambuStudio #7144: slow launch of bambu studio

mimix1004, 2025-06-05:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 64bits

### Additional system information

ryzen 7 5700X3D, 80GO ram, RTX3700, NVME 500Go

### Printer

X1C

### How to reproduce

launch bambu studio

### Actual results

take 4/5 minutes to access of home, and home is empty.
I can access after that to the prepare menu and do what i want

### Expected results

rapid launch exactly same as before the recently update

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-06-17:
It might have been an occasional cloud service issue. Is it working properly now? Can you access the MakerWorld website normally using a browser?

JohnSmithCYM, 2025-08-14:
Can you reproduce this problem with https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85?