bambulab/BambuStudio #7147: Dosen't change nozzle setting

asher8554, 2025-06-05:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

장치 이름	DESKTOP
프로세서	12th Gen Intel(R) Core(TM) i7-12700   2.10 GHz
설치된 RAM	32.0GB(31.7GB 사용 가능)
시스템 종류	64비트 운영 체제, x64 기반 프로세서
펜 및 터치	펜과 2개 터치 포인트 지원

### Printer

A1 mini

### How to reproduce

[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jcollasius, 2025-06-06:
Welcome to the club. Pull up a chair and watch the programmers as they fail to understand the problem but are also unable to explain the new workflow for changing the nozzle.
I don't understand a word of Chinese (if that is Chinese...), but I recognise the dialogue and am facing the same problem.
At least both the old version and Orca Slicer work...

JohnSmithCYM, 2025-06-12:
Thank you for the feedback. In version V2.1.0, we disabled the nozzle diameter and type settings for all printer models (starting from V2.0.3, we had already disabled these settings for the H2D), because we believe that this operation should only be performed by the user physically present at the printer.

eried, 2025-06-12:
> Thank you for the feedback. In version V2.1.0, we disabled the nozzle diameter and type settings for all printer models (starting from V2.0.3, we had already disabled these settings for the H2D), [...]

Can you please let us at least trigger the nozzle identification (on H2D) remotely?