bambulab/BambuStudio #7148: [BUG] "Value out of range" dialog box won't close (MacOS)

bolsoncerrado, 2025-06-05:
### Bambu Studio Version

2.0.1.5.9

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Somona

### Additional system information

M1Ultra

### Printer

N/A

### How to reproduce

1. Go to GLOBAL settings
2. SUPPORT tab
3. ADVANCED
4. Enable INITIAL LAYER DESING and add a 90

### Actual results

A popup warning will appear and 90 will be replaced by a 10. You can't close the dialog no matter what.

### Expected results

Close the dialog box and keep working as expected.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-06-09:
Hello, is there any prerequisite? We used the default blank project and did not reproduce it~

Haidiye00, 2025-06-09:
Hello, can your 3mf still be stably reproduced by reopening the software? If it can be stably reproduced, could you provide this 3mf? Thank you

closed 2026-11-03 by Haidiye00