bambulab/BambuStudio #7158: Problem slicing flush objects

seanbo, 2025-06-06:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu A1 Mini

### How to reproduce

1) Place one object on top of another and merge them
2) Set the color of the objects to the desired color
3) Press slice

[image]
[image]

### Actual results

In the attached 3mf file, you will see that the "F" by itself slices fine.  You will see that the "F" raised above the level of the other object (far right) slices fine.  You will see that the "flush F" is not sliced (center part).

### Expected results

I expect the purple F to show.  I have tried changing the print order as well as the object order and neither way works.

### Project file & Debug log uploads

[MW-TunnelsSCAD.zip](https://github.com/user-attachments/files/20624016/MW-TunnelsSCAD.zip)

[debug_Fri_Jun_06_03_32_39_6436.log.zip](https://github.com/user-attachments/files/20624044/debug_Fri_Jun_06_03_32_39_6436.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-06-13:
Thanks for reporting this issue, we will check it.

seanbo, 2025-06-26:
> [@seanbo](https://github.com/seanbo) When the faces of two parts overlap, the painting color takes priority. The top of your model is painted black, so it is displayed as black. I cleared all the painting information, and saved the file. You can try it.
> 
> [MW-TunnelsSCAD-save-as.zip](https://github.com/user-attachments/files/20927634/MW-TunnelsSCAD-save-as.zip)

I didn't even realize it was painted.  Thank you!