bambulab/BambuStudio #7160: De-retraction length

mszulen, 2025-06-06:
**Is your feature request related to a problem? Please describe.**
When printing LW-PLA (sometimes also others) it is advised/required to modify the de-retraction length.
This is easily done with other slicers (Prusa/SuperSlicer/Orca/Cura) but not Bambu studio (which only has de-retraction speed parameter available).
 It's especially annoying now, when we can't use Orca.

**Describe the solution you'd like**
Add a de-retraction length parameter in the filament overrides

**Describe alternatives you've considered**
Allow OrcaSlicer to work with Bambu printers like before.

Here's screenshot from Orca:

[image]

KenPNT, 2025-07-18:
Was looking for this feature on BS. Please add this! For the time being, I wonder if there's any equivalent solution?

mszulen, 2025-08-14:
> Was looking for this feature on BS. Please add this! For the time being, I wonder if there's any equivalent solution?

Yes, use other slicer...... :)

mszulen, 2025-10-17:
I see it has been added to the recent beta. Thank you!