bambulab/BambuStudio #7163: Cameratools won't download

mctazman, 2025-06-06:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS v

### Additional system information

M4, 32GB RAM, When trying to open livestream, the cameratools won't download. BS has full access in the firewall, no VPN present.

### Printer

X1c

### How to reproduce

Click Go Live

### Actual results

[image]
[image]

### Expected results

Cameratools to be installed

### Project file & Debug log uploads

[image]

[log_iotc.txt](https://github.com/user-attachments/files/20625081/log_iotc.txt)

[Archief.zip](https://github.com/user-attachments/files/20625103/Archief.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

adamscybot, 2025-06-07:
I've got this too. I found the relevant log lines:

```
2025-06-07 20:32:01.417126[Thread 0x00002684]:[UpgradeNetworkJob process]: enter
2025-06-07 20:32:01.417126[Thread 0x00002684]:UpgradeNetworkJob: save netowrk_plugin to C:\Users\adam\AppData\Local\Temp\camera_tools.zip.51040.tmp
2025-06-07 20:32:01.417126[Thread 0x00002684]:[download_plugin]: enter
2025-06-07 20:32:01.417126[Thread 0x00002684]:[download_plugin]: check the plugin from https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/cameratools/cloud=02.01.00.00
2025-06-07 20:32:01.513184[Thread 0x000057f8]:extrusion_cali_get: {"print":{"command":"extrusion_cali_get","filament_id":"","nozzle_diameter":"0.4","sequence_id":"20012"}}
2025-06-07 20:32:01.516186[Thread 0x000057f8]:publish_json: {"print":{"command":"extrusion_cali_get","filament_id":"","nozzle_diameter":"0.4","sequence_id":"20012"}} code: 0
2025-06-07 20:32:01.604696[Thread 0x00002684]:[download_plugin 1]: no available plugin found for this app version: 02.01.00.59
```

The Bambu API is indeed not returning an entry for camera tools on the URL it hits.

adamscybot, 2025-06-09:
Nice, in that case a shortcut workaround might be to just download the [2.0 tarball](https://public-cdn.bblmw.com/upgrade/studio/software/02.01.00.59/06fd96547f/Bambu_Studio_win-v02.01.00.59.exe) and extract it manually into the plugins folder (on windows i think somewhere in %appdata% or %localappdata%)

furynick, 2025-06-09:
> Nice. Small correction -- I think you meant "get back to" 2.1 not 2.01.
You're right, I just copied version from api response.

closed 2025-07-01 by BoBoYu-BamBu