bambulab/BambuStudio #7168: Crashing on every print job

gmcfad, 2025-06-06:
### Bambu Studio Version

2.0.1.59

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 15.5

### Printer

Bambu Labs A1

### How to reproduce

1. Send job to printer
2. it crashes after sending every job today

### Actual results

Crashing

### Expected results

Fix crashing issue

### Project file & Debug log uploads

[debug_Fri_Jun_06_09_12_52_915.log.0.zip](https://github.com/user-attachments/files/20632983/debug_Fri_Jun_06_09_12_52_915.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gmcfad, 2025-06-10:
Here is the dump…….

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [989]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               02.01.00.59 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-06-10 07:47:50.7389 -0400
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        80C38513-6803-32BD-C5EF-C33DB0F48FDF

Sleep/Wake UUID:       ECA64C7D-3ADD-44F5-B71E-C385FBB4C17D

Time Awake Since Boot: 370 seconds
Time Since Wake:       3 seconds

System Integrity Protection: enabled

Crashed Thread:        30

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [989]

VM Region Info: 0 is not in any region.  Bytes before following region: 4336386048
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102780000-106960000    [ 65.9M] r-x/r-x SM=COW  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x1028e6684 nlohmann::detail::serializer<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>>::dump_escaped(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool) + 216
1   BambuStudio                   	       0x1028e5b78 nlohmann::detail::serializer<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>>::dump(nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>> const&, bool, bool, unsigned int, unsigned int) + 2200
2   BambuStudio                   	       0x102e4f770 nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>::dump(int, char, bool, nlohmann::detail::error_handler_t) const + 556
3   BambuStudio                   	       0x103af9a20 Slic3r::MachineObject::parse_json(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool) + 82080
4   BambuStudio                   	       0x10368d978 wxAsyncMethodCallEventFunctor<Slic3r::GUI::GUI_App::init_networking_callbacks()::$_20::operator()(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) const::'lambda'()>::Execute() + 660
5   BambuStudio                   	       0x105c71960 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
6   BambuStudio                   	       0x105c71730 wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   BambuStudio                   	       0x105c71580 wxEvtHandler::ProcessPendingEvents() + 236
8   BambuStudio                   	       0x105bd586c wxAppConsoleBase::ProcessPendingEvents() + 112
9   BambuStudio                   	       0x105c7a43c wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
10  CoreFoundation                	       0x18f2e9098 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
11  CoreFoundation                	       0x18f2e8f80 __CFRunLoopDoObservers + 536
12  CoreFoundation                	       0x18f2e85f0 __CFRunLoopRun + 784
13  CoreFoundation                	       0x18f2e7c58 CFRunLoopRunSpecific + 572
14  HIToolbox                     	       0x19ad7c27c RunCurrentEventLoopInMode + 324
15  HIToolbox                     	       0x19ad7f31c ReceiveNextEventCommon + 216
16  HIToolbox                     	       0x19af0a484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
17  AppKit                        	       0x19320fab4 _DPSNextEvent + 684
18  AppKit                        	       0x193bae5b0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
19  AppKit                        	       0x193202c64 -[NSApplication run] + 480
20  BambuStudio                   	       0x105ba1ea0 wxGUIEventLoop::OSXDoRun() + 112
21  BambuStudio                   	       0x105c7ab20 wxCFEventLoop::DoRun() + 40
[... cut ...] ymbolLocation":136,"imageIndex":14},{"imageOffset":7040,"symbol":"thread_start","symbolLocation":8,"imageIndex":14}],"threadState":{"x":[{"value":260},{"value":0},{"value":0},{"value":0},{"value":0},{"value":160},{"value":0},{"value":99997959},{"value":6191491976},{"value":0},{"value":0},{"value":2},{"value":2},{"value":0},{"value":0},{"value":0},{"value":305},{"value":8559910752},{"value":0},{"value":6191492264},{"value":6191492328},{"value":6191493344},{"value":99997959},{"value":0},{"value":0

JohnSmithCYM, 2025-06-25:
We  are sorry for that, have you tried V2.1.1 and it's new network plugin?

gmcfad, 2025-06-25:
> JohnSmithCYM

I’m running 2.1.1.52.

Is it possible this is an issue with “sleep mode” on the laptop.

I’ve never seen it crash - it always crashes after I kick a print off and then come back later to either check on it or start another print.

Gene

XinZhangBBL, 2025-07-21:
Halo, we have solved the problem at latest release. please check https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60。

[image]

gmcfad, 2025-07-26:
> XinZhangBBL
>  left a comment
> (bambulab/BambuStudio#7168)
> Halo, we have solved the problem at latest release. please check https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60。 <https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60%E3%80%82>image.png (view on web) <https://github.com/user-attachments/assets/cb24766c-a271-4aeb-a4fa-5a6b1572434c>

From today’s crash - with new software…..

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [1479]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               02.02.00.60 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-07-26 13:18:47.8887 -0400
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        80C38513-6803-32BD-C5EF-C33DB0F48FDF

Sleep/Wake UUID:       8465C996-0C3C-46E4-BF7E-93962E412F34

Time Awake Since Boot: 11000 seconds
Time Since Wake:       55 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   BambuStudio [1479]

Application Specific Information:
abort() called

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19e1a9388 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19e1e288c pthread_kill + 296
2   libsystem_c.dylib             	       0x19e0ebc60 abort + 124
3   libc++abi.dylib               	       0x19e19839c abort_message + 132
4   libc++abi.dylib               	       0x19e186cf0 demangling_terminate_handler() + 316
5   libobjc.A.dylib               	       0x19de0cde4 _objc_terminate() + 172
6   libc++abi.dylib               	       0x19e1976b0 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x19e197654 std::terminate() + 108
8   libbambu_networking.dylib     	       0x113362440 0x112ffc000 + 3564608
9   libbambu_networking.dylib     	       0x1132f0e80 0x112ffc000 + 3100288
10  libsystem_c.dylib             	       0x19e09a944 __cxa_finalize_ranges + 480
11  libsystem_c.dylib             	       0x19e09a704 exit + 44
12  libdyld.dylib                 	       0x19e1ecdc8 dyld4::LibSystemHelpers::exit(int) const + 20
13  dyld                          	       0x19de42cac dyld4::LibSystemHelpersWrapper::exit(int) const + 172
14  dyld                          	       0x19de42bc8 start + 6124

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	       0x19e1a0c34 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x19e1b33a0 mach_msg2_internal + 76
2   libsystem_kernel.dylib        	       0x19e1a9764 mach_msg_overwrite + 484
3   libsystem_kernel.dylib        	       0x19e1a0fa8 mach_msg + 24
4   CoreFoundation                	       0x19e2cde7c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x19e2cc798 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x19e2cbc58 CFRunLoopRunSpecific + 572
7   AppKit                        	       0x1a23177fc _NSEventThread + 140
8   libsystem_pthread.dylib       	       0x19e1e2c0c _pthread_start + 136
9   libsystem_pthread.dylib       	       0x19e1ddb80 thread_start + 8

Thread 2:: JavaScriptCore libpas scavenger
0   libsystem_kernel.dylib        	       0x19e1a43cc __psynch_cvwait + 8
1   libsystem_pthread.dylib       	       0x19e1e30e0 _pthread_cond_wait + 984
2   JavaScriptCore                	       0x1be6f2250 scavenger_thread_main + 1584
[... 494 lines cut ...]
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,1, BootROM 11881.121.1, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Mar 23 2025 19:56:28 version 20.130.17.0.8.7.197 FWID 01-764e34b7
IO80211_driverkit-1475.39 "IO80211_driverkit-1475.39" Apr 18 2025 20:10:40
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

Gene McFadden

XinZhangBBL, 2025-07-28:
Could you reproduce the problem to offer log for us. you can refer to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log. Thank you very much for your cooperation.