bambulab/BambuStudio #7170: warning staying on screen

raatzy, 2025-06-06:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

it's across all platforms, bamby handy, laptop, etc

[image]

[image]

### Printer

Bambu Lab X1C

### How to reproduce

no need to reproduce - the warning never goes away. X1C printer only.

### Actual results

Nothing. It always stays the same.

### Expected results

Would like it to be fixed.

### Project file & Debug log uploads

[debug_Fri_Jun_06_19_41_42_34060.log.zip](https://github.com/user-attachments/files/20635257/debug_Fri_Jun_06_19_41_42_34060.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-10:
@raatzy Hello, if the consumables or spools of your AMS get stuck and cannot be pulled back to the AMS from the tool head, this warning will appear; Solution: You can go to the machine and check if there are any consumables that haven't been returned successfully in your AMS slot. Manually return them, and the warning will disappear.

raatzy, 2025-06-18:
> @raatzy <https://github.com/raatzy> Have you solved your problem yet?

Yes. Fixed it last night. I just took out the filament and loaded a new
roll of filament in a different colour.

So yes. Solved.

Close job.

Thanks for checking.

closed 2025-06-18 by BoBoYu-BamBu