bambulab/BambuStudio #7172: Painted brims lost after saving / reopening

blaskotron, 2025-06-07:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

9950x3d 6900xt 32gb ram 

### Printer

H2D 

### How to reproduce

1. Add painted brims to this object. 
2. Save file. 
3. Open another file. 
4. Reopen painted brims file. 

### Actual results

Brims are gone on reopen. 

### Expected results

Brim should still be there.

### Project file & Debug log uploads

[Ulilac Pocket Clip ASA (2).zip](https://github.com/user-attachments/files/20636464/Ulilac.Pocket.Clip.ASA.2.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

MadMax389, 2025-07-13:
I can confirm this bug, however it only happens to me with the following conditions: 
1. Item is in an assembly
2. Only happens with manually painted brims. Auto-generated brims continue to work after re-opening the project.

blaskotron, 2025-10-26:
> This can be confirmed as a bug. Thank you for your feedback.

Any update on this? Still have to repaint the brims every time I use this file.