bambulab/BambuStudio #7176: Bambu Studio Crashes Every Time I Select Variable Layer Height

Heuristicmystic, 2025-06-07:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	Snapdragon(R) X Elite - X1E80100 - Qualcomm(R) Oryon(TM) CPU   3.42 GHz
Installed RAM	16.0 GB (15.6 GB usable)
Device ID	CCC059B6-F5FA-484B-9F54-08F399F47C49
Product ID	00342-21257-65178-AAOEM
System type	64-bit operating system, ARM-based processor

### Printer

X1C

### How to reproduce

Select an object and click on Variable layer height

### Actual results

Bambu Studio disappears 

### Expected results

Bambu Studio doesn't crash and enables variable layer height

### Project file & Debug log uploads

The project can't be recovered once Variable layer height is clicked

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-06-09:
[image]
It seems works well with my Studio on Win11 and can you offer your dmp file for us to analyze? You can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get dmp file.

Heuristicmystic, 2025-06-13:
Got it!
[crash_Fri_Jun_13_10_17_05_0.log](https://github.com/user-attachments/files/20728116/crash_Fri_Jun_13_10_17_05_0.log)

Heuristicmystic, 2025-06-19:
Can you clarify?