bambulab/BambuStudio #7177: Bambu lab studio

neseu, 2025-06-07:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Printer

BAMBU LAB P1s

### How to reproduce

bambu lab studio
model online 
search something
erease serching word

### Actual results

browser in Model online tab is not searching during the blank field after prior serarch required model

### Expected results

while search field is blank, should searching default models as on start app

### Project file & Debug log uploads

no projevt

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-10:
@neseu Thank you for your feedback. this page is embedded within the studio software in the mall. Therefore, all issues related to this page will be handled by the mall side.
[image]

BoBoYu-BamBu, 2025-06-10:
@neseu  I have already informed the mall team and they will make the necessary changes. Thank you.