bambulab/BambuStudio #7182: The painting function does not work for negative parts

msart0, 2025-06-07:
### Bambu Studio Version

2.1.0.59 

### Where is the application from?

Bambu Lab Official website

### OS version

mac 15.5 (24F74)

### Additional system information

Apple M1 Pro  
Macbook Pro 2021

### Printer

H2D

### How to reproduce

In the preparation interface, select coloring, coloring by layer, and then slicing. Observe the slicing results, from the 4th to the 25th layer are all painted black.

However, after adding the negative part (2 half holes, for placing the magnet after printing), the negative part area is restored to the color before coloring.

The number of color changes and printing time have increased a lot.

Before and after adding the negative part, coloring by layer cannot change the color around the negative part.

[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[44.zip](https://github.com/user-attachments/files/20642117/44.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

msart0, 2025-06-07:
Expected effect: After adding a negative shape, coloring will take effect around the negative shape.

Haidiye00, 2025-06-11:
Hello, after internal analysis, coloring can only be applied to a single mesh and cannot be applied to the combined effect of multiple meshes.
So it is recommended that you first use Boolean tools, perform Boolean operations, and then color them to take effect.

I will close this question for now. If there are still issues, let's continue discussing

closed 2025-06-11 by Haidiye00