bambulab/BambuStudio #7187: Big inconvenience in the new version of bamboo studio

Vlad-313, 2025-06-08:
In The old version of the program doesn't have this useless border

[image]

why and what for is it present in new versions ?

[image]

it interferes with the precise fitting of parts.

pfeerick, 2025-06-09:
I would suggest you change the title of this issue from "Big inconvenience in the new version of bamboo studio" (what is? why should I care to read the issue?) to something like "Object outline interferes with part alignment" or something similar. 

The online was there with prior versions when you initially select the part, but went away once you clicked the move icon / pressed the keyboard shortcut. For me, I'm seeing the outline after selecting the Move option on Linux Mint 22, using the 24.04 AppImage with 2.1.0 (but didn't with 2.0.x). I personally consider this to be a bug rather than feature/enhancement request.

Vlad-313, 2025-06-09:
This was not a problem until I could use older versions of Bamboo Studio that did not have this useless outline. But the latest firmware of the printer does not allow using the older version of Bamboo Studio.

eried, 2025-06-12:
The outline should be very transparent (20% opaque) or maybe be just a glow around