bambulab/BambuStudio #7195: Exhaust/Cooling Fan Should be controllable no matter what material is used.

danielnc06, 2025-06-08:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Start an ABS print, goto Device Tab, try to adjust the exhaust fan. 

### Actual results

[image]

It will not allow you to adjust exhaust fan when printing ABS. I understand why and you want chamber temperatures up with ABS to prevent warping. Sure.. But I use a low 40% exhaust fan to pull a negative pressure and suck ABS fumes out the window with ducting. 

I cannot turn this on via Bambu Studio anymore. There can be a warning but I need to be able to do this. 

### Expected results

Click Fans while printing ABS. get a warning, and then be able to turn on whatever I want manually.

### Project file & Debug log uploads

Not needed. Any ABS print will do this. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vanowm, 2025-06-08:
Can you control fan via printer's display? (also, what firmware your printer is running?)

changbaishen, 2025-06-11:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]