bambulab/BambuStudio #7205: Slicer crashes all the time

jontimi, 2025-06-09:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz   3.19 GHz
32.0 GB RAM
NVIDIA QUADRO P600

### Printer

Bambu lab P1S

### How to reproduce

Open studio
slice stl
(sometimes it crashes now)
change color scheme in preview
(crashes alot here)
send to print 
(crashes before sending alot)
it wasnt like this 2 weeks ago... i guess bad update 

### Actual results

studio crashes
when i reopen it , it suggests autorecover.
but keeps crashing at the same stages

### Expected results

stop crashing? sending crash report?

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20651172/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-06-10:
@jontimi Sorry, could you please provide the 3mf file?

Haidiye00, 2025-06-10:
[image]
Hello，If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

jontimi, 2025-06-10:
thanks, just did the change to true. seems to work. ill update if it crashes again

Haidiye00, 2025-06-10:
Okay, thank you for your feedback. We will continue to upgrade the rendering of slices in the future~