bambulab/BambuStudio #7213: llvm-libs 20+ upgrade affects Bambu Studio

Nordaar, 2025-06-09:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Built from source by yourself or others, Bambu Lab github releases

### OS version

linux 6.14.10.arch1-1

### Additional system information

ArchLinux, Wayland, Hyprland, AUR package bambustudio-nvidia-bin 02.01.00.59-1 (https://aur.archlinux.org/packages/bambustudio-nvidia-bin) which runs the AppImage release (https://github.com/bambulab/BambuStudio/releases/download/v02.01.00.59/Bambu_Studio_linux_fedora-v02.01.00.59.AppImage)

### Printer

H2D

### How to reproduce

With the recent upgrade to `llvm-libs` from `19.1.7-2` to `20.1.6-3` , when running bambu-studio an error is printed: `/opt/bambustudio-bin/bin/bambu-studio: error while loading shared libraries: libLLVM.so.19.1: cannot open shared object file: No such file or directory`.
Downgrading the `llvm-libs` back to `19.7.7-2` fixes the issue and `bambu-studio` runs normally.

### Actual results

[image]

### Expected results

included above

### Project file & Debug log uploads

`/opt/bambustudio-bin/bin/bambu-studio: error while loading shared libraries: libLLVM.so.19.1: cannot open shared object file: No such file or directory`

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Nordaar, 2025-06-10:
For Arch+nVidia users:

the AUR `bambustudio-nvidia-bin 02.01.00.59-2` package now comes bundled with llvm libs and works out of the box. 

tested with:

- `linux 6.14.10.arch1-1`
- `bambustudio-nvidia-bin 02.01.00.59-2`
- `mesa 1:25.1.3-3`
- `wayland 1.23.1-2`
- `hyprland 0.49.0-1`
- `nvidia-dkms 575.57.08-1`
- `llvm-libs 20.1.6-3`

MackBambu, 2025-06-17:
The Flatpak version packages all the runtime environments, making it more stable to use.
https://flathub.org/apps/com.bambulab.BambuStudio