bambulab/BambuStudio #7214: Bambu Studio Crashes in Mac too often

kazthehack, 2025-06-09:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Mac 15.5, Apple M4

### Additional system information

Memory 32 Gb

### Printer

A1

### How to reproduce

Happens in different scenarios 

- Adding a new custom filament 
- Configuring a new custom filament
- Assigning the custom filament in the device tab
- Trying to print with the custom filament after slicing

-------
After a couple of crashes, there are cases where in the project file deletes the custom filament and is not recoverable. 

### Actual results

[image]

### Expected results

Should not crash

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/20662048/Archive.zip)

[Logs.zip](https://github.com/user-attachments/files/20662054/Logs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

kazthehack, 2025-06-10:
[Archive.zip](https://github.com/user-attachments/files/20666942/Archive.zip)

forgot to add the logs generated from the video

kazthehack, 2025-06-11:
i have multiple printers btw, and within print file, i swap and select to which printer i want to send the file to

DaPrint, 2025-06-12:
Can confirm, got an A1 and now it crashes seemingly whenever a print finishes...

harry-bbl, 2025-07-23:
@kazthehack need studio log to analyze, you can find log folder by clicking "help->Show Configuration Folder"

kazthehack, 2025-07-24:
Hi its me, i dont seem to replicate it as much these days. Looks like its fixed. You can now close the issue